tasmota/docker-tasmota #11: fatal error: IRremoteESP8266.h: No such file or directory

raxod502, 2022-08-22:
Got the following error using the Docker image to build Tasmota:

<details>

```
% sudo docker run -it --rm -v $PWD:/tasmota -u $UID:$GID blakadder/docker-tasmota -e tasmota
Unable to find image 'blakadder/docker-tasmota:latest' locally
latest: Pulling from blakadder/docker-tasmota
c7b7d16361e0: Pull complete 
b7a128769df1: Pull complete 
1128949d0793: Pull complete 
667692510b70: Pull complete 
bed4ecf88e6a: Pull complete 
8a8c75f3996a: Pull complete 
59ba7226fec9: Pull complete 
8f449aa0129e: Pull complete 
6e376b1610c0: Pull complete 
c881dc93176e: Pull complete 
702e62e4b262: Pull complete 
c1abbae766a9: Pull complete 
98034b5d313f: Pull complete 
5c77f8c58c8a: Pull complete 
Digest: sha256:2264a2ea02f65aa440e3ba3b624ea2a43b82881f3c4ec14be1b4dc621cdad593
Status: Downloaded newer image for blakadder/docker-tasmota:latest
Compiling Tasmota...
Processing tasmota (platform: https://github.com/tasmota/platform-espressif8266/releases/download/v2.7.4.9/platform-espressif8266-2.7.4.9.zip; framework: arduino; board: esp8266_1M)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing https://github.com/tasmota/platform-espressif8266/releases/download/v2.7.4.9/platform-espressif8266-2.7.4.9.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif8266@2.7.4+9 has been installed!
Tool Manager: Installing platformio/toolchain-xtensa @ ~2.40802.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%
Tool Manager: toolchain-xtensa@2.40802.200502 has been installed!
Tool Manager: Installing tasmota/framework-arduinoespressif8266 @ ~2.7.4
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoespressif8266@2.7.4+9 has been installed!
Tool Manager: Installing https://github.com/tasmota/esptool/releases/download/v3.3.1/esptool-3.3.1.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.30301.220515 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8266_1M.html
PLATFORM: Espressif 8266 (2.7.4+9) > Espressif Generic ESP8266 Tasmota 1M sketch NO FS
HARDWARE: ESP8266 80MHz, 80KB RAM, 972.00KB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 2.7.4+9 
 - tool-esptoolpy @ 1.30301.220515 (3.3.1) 
 - toolchain-xtensa @ 2.40802.200502 (4.8.2)
Converting tasmota.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- headers
|   |-- EEPROM @ 1.0
|   |-- ESP8266WebServer @ 1.0
|   |   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- EEPROM 24C128_256_521
|   |-- Wire @ 1.0
|-- UdpListener @ 1.0
|-- Unishox Compressor Decompressor highly customized and optimized for ESP8266 and Tasmota @ 1.0
|-- TLS mini client derived from Arduino @ 1.0
|   |-- ESP8266WiFi @ 1.0
|   |-- BearSSL @ 0.6
|-- DNSServer @ 1.1.1
|   |-- ESP8266WiFi @ 1.0
|-- DnsClient @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- ESP8266HTTPClient @ 1.2
|   |-- ESP8266WiFi @ 1.0
|-- ESP8266WebServer @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- ESP8266WiFi @ 1.0
|-- ESP8266httpUpdate @ 1.3
|   |-- ESP8266HTTPClient @ 1.2
|   |   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|   |-- ESP8266WiFi @ 1.0
|-- JSMN JSON parser customized and optimized for ESP8266 and Tasmota @ 1.0
|-- TasmotaLList @ 1.0
|-- LinkedList
|-- LittleFS(esp8266) @ 0.1.0
|-- PubSubClient @ 2.8
|-- Ext-printf @ 1.0
|-- SD(esp8266) @ 2.0.0
|   |-- SDFS @ 0.1.0
|   |   |-- SPI @ 1.0
|   |   |-- ESP8266SdFat @ 1.1.0
|   |   |   |-- SPI @ 1.0
|-- SPI @ 1.0
|-- ESP8266SdFat @ 1.1.0
|   |-- SPI @ 1.0
|-- TasmotaSerial @ 3.5.0
|-- Ticker @ 1.0
|-- Wire @ 1.0
|-- base64 @ 1.1.1
|-- BearSSL @ 0.6
Building in release mode
*** use provided user_config_override.h as planned ***
Using manually specified: COM5
Compiling .pio/build/tasmota/src/tasmota.ino.cpp.o
Compiling .pio/build/tasmota/src/tasmota_xsns_sensor/xsns_62_esp32_mi_homekit.c.o
Compiling .pio/build/tasmota/src/tasmota_support/homekit.c.o
Generating LD script .pio/build/tasmota/ld/local.eagle.app.v6.common.ld
Compiling .pio/build/tasmota/lib42e/EEPROM/EEPROM.cpp.o
Compiling .pio/build/tasmota/libbc6/ESP8266WiFi/BearSSLHelpers.cpp.o
In file included from tasmota/my_user_config.h:1142:0,
                 from /tasmota/tasmota/tasmota.ino:27:
tasmota/user_config_override.h:11:0: warning: "MQTT_TLS_ENABLED" redefined [enabled by default]
 #define MQTT_TLS_ENABLED true
 ^
In file included from /tasmota/tasmota/tasmota.ino:27:0:
tasmota/my_user_config.h:180:0: note: this is the location of the previous definition
 #define MQTT_TLS_ENABLED       false             // [SetOption103] Enable TLS mode (requires TLS version)
 ^
Compiling .pio/build/tasmota/libbc6/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling .pio/build/tasmota/libbc6/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling .pio/build/tasmota/libbc6/ESP8266WiFi/ESP8266WiFiAP.cpp.o
Compiling .pio/build/tasmota/libbc6/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
Compiling .pio/build/tasmota/libbc6/ESP8266WiFi/ESP8266WiFiGratuitous.cpp.o
/tasmota/tasmota/tasmota_xdrv_driver/xdrv_05_irremote.ino:62:29: fatal error: IRremoteESP8266.h: No such file or directory

*************************************************************************
* Looking for IRremoteESP8266.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:IRremoteESP8266.h"
* Web  > https://registry.platformio.org/search?q=header:IRremoteESP8266.h
*
*************************************************************************

 #include <IRremoteESP8266.h>
                             ^
compilation terminated.
Compiling .pio/build/tasmota/libbc6/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Compiling .pio/build/tasmota/libbc6/ESP8266WiFi/ESP8266WiFiSTA-WPS.cpp.o
Compiling .pio/build/tasmota/libbc6/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
Compiling .pio/build/tasmota/libbc6/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Compiling .pio/build/tasmota/libbc6/ESP8266WiFi/WiFiClient.cpp.o
*** [.pio/build/tasmota/src/tasmota.ino.cpp.o] Error 1
================================================================================ [FAILED] Took 16.82 seconds ================================================================================

Environment    Status    Duration
-------------  --------  ------------
tasmota        FAILED    00:00:16.821
=========================================================================== 1 failed, 0 succeeded in 00:00:16.821 ===========================================================================
All done! Find your builds in Tasmota/build_output/firmware/
```

</details>

This is with Tasmota v12.1.0 (806c5762bc19b322bfa3634e730fa11f89df823a) and the following `user_config_override.h`:

```c
/*
  user_config_override.h - user configuration overrides my_user_config.h for
  Tasmota
*/

#ifndef _USER_CONFIG_OVERRIDE_H_
#define _USER_CONFIG_OVERRIDE_H_

#ifndef USE_MQTT_TLS
#define USE_MQTT_TLS
#define MQTT_TLS_ENABLED true
#define USE_MQTT_TLS_CA_CERT
#endif

#endif // _USER_CONFIG_OVERRIDE_H_
```

Interestingly, the message `All done! Find your builds in Tasmota/build_output/firmware/` is printed despite the immediately preceding message `tasmota        FAILED    00:00:16.821`.

raxod502, 2022-08-22:
Interesting, there is such a file in the Tasmota repository I am using:

```
% file lib/lib_basic/IRremoteESP8266/IRremoteESP8266/src/IRremoteESP8266.h
lib/lib_basic/IRremoteESP8266/IRremoteESP8266/src/IRremoteESP8266.h: C source, Unicode text, UTF-8 text
```

blakadder, 2022-08-23:
that's a Tasmota compilation error not this docker container. seek help in Tasmota support channels

closed 2022-08-23 by blakadder