tasmota/docker-tasmota #14: ARM64 container

isarrider, 2023-01-10:
Hi,

it would be awesome to add an ARM64 container to docker hub...
Is this possible?

BR,
Alex

barbudor, 2023-01-10:
Why not just cloning the repo and building the image on your system?

isarrider, 2023-01-10:
because so everybody can profit... ;)

barbudor, 2023-01-10:
I believe you need an ARM64 platform to create the image
So blakadder doesn't have one or if github action doesn't provide such option, then it's may not be t be possible
I don't know if it is possible to cross build a docker image and I believe it's probably not possible because you need to run code such as 'apt install' in the target os

isarrider, 2023-01-10:
it would be, e.g. with
`docker buildx build --platform linux/arm64,linux/arm/v7 -t docker-tasmota`
for arm64 and armv7 in the example above...
But I guess github actions build them anyway?

blakadder, 2023-01-10:
i have no intention on adding it, container is build on my x86 machine and pushed to docker

closed 2023-01-11 by isarrider

Jason2866, 2024-06-03:
added linux/arm64. There are no espressif toolchains available for arm/v7