tasmota/docker-tasmota #15: Error: Unknown board ID 'esp32_4M'

jeroenst, 2023-01-19:
When I'm building tasmota32 with blakadder/docker-tasmota I get the error Error: Unknown board ID 'esp32_4M'

```
Processing tasmota32 (framework: arduino; platform: https://github.com/tasmota/platform-espressif32/releases/download/2022.12.2/platform-espressif32.zip; board: esp32_4M)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing https://github.com/tasmota/platform-espressif32/releases/download/2022.12.2/platform-espressif32.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@2022.12.2 has been installed!
Error: Unknown board ID 'esp32_4M'
```

How can I solve this?

barbudor, 2023-01-19:
I've alerady answered on tasmota
I can't reproduce
You may have an old override file around

jeroenst, 2023-01-21:
I removed all old files and downloaded them again. Now everything works fine. Thank you.

closed 2023-01-21 by jeroenst