tasmota/docker-tasmota #16: How to update docker image to last platformio

blacknell, 2023-03-14:
I'm getting errors below
```
Error: Development platform 'espressif32' is not compatible with PlatformIO Core v6.1.5 and depends on PlatformIO Core >=6.1.6.
```
As far as I can tell the PlatformIO library needs updating inside the image and the compile script seems to imply it will update this.

Not sure how to force it and why it doesn't refresh with a new docker pull

Any pointers appreciated

blakadder, 2023-03-14:
platformio will upgrade if you build the container locally

blacknell, 2023-03-14:
Ok. 

Executing `docker run -ti --rm -v $(pwd)/Tasmota:/tasmota -u $UID:$GID docker-tasmota -e tasmota32`

gives me

```
Error: Invalid '/tasmota/platformio.ini' (project configuration file): 'No section: 'core32solo1''
```
I can't find any reference to this in the my previous set up

closed 2023-03-15 by blacknell