tasmota/docker-tasmota #17: features arent compiled in builds

Gifford47, 2023-03-14:
hi!

i want to compile (on debian) my own build with multiple feature enabled like shutters or sensors. but my builds do not contain these features. i tested several builds with shutter feature.

platformio_override.ini and user_config_override.h are copied to docker-tasmota/Tasmota.

platformio_override.ini:
```
[env:tasmota_latest-full_rules]
build_unflags           = ${common.build_unflags}
build_flags             = ${env.build_flags} -D FW_FULLRULES

[env:tasmota_latest_sensors-shutter-full_rules]
;extends = env:tasmota-sensors
build_unflags           = ${common.build_unflags}
build_flags             = ${env.build_flags} -D FW_FULLRULES -D SHUTTERS -D SENSORS

[env:tasmota_latest_lite-full_rules]
;extends = env:tasmota-lite
build_unflags           = ${common.build_unflags}
build_flags             = ${env.build_flags} -D FW_FULLRULES -D LITE

[env:tasmota_latest_lite-shutter-full_rules]
;extends = env:tasmota-lite
build_unflags           = ${common.build_unflags}
build_flags             = ${env.build_flags} -D FW_FULLRULES -D SHUTTERS -D LITE

[env:tasmota_latest_min-full_rules]
;extends = env:tasmota-minimal
build_unflags           = ${common.build_unflags}
build_flags             = ${env.build_flags} -D FW_FULLRULES -D MIN
```

user_config_override.h:
```
#ifdef SHUTTERS
    #ifndef USE_SHUTTER
    #define USE_SHUTTER             // Add Shutter support for up to 4 shutter with different motortypes (+6k code)
    #define SHUTTER_CLEAR_PWM_ONSTOP    // for shutter pwm on endstop
    #endif
#endif

#ifdef FW_FULLRULES
    // -- CODE_IMAGE_STR is the name shown between brackets on the 
    //    Information page or in INFO MQTT messages
    #undef CODE_IMAGE_STR
    #define CODE_IMAGE_STR "fullrules"
    
    #ifndef USE_EXPRESSION
    #define USE_EXPRESSION         // Add support for expression evaluation in rules (i.e. =,+,-, ...) (+3k2 code, +64 bytes mem)  
    #endif

    #ifndef SUPPORT_IF_STATEMENT
    #define SUPPORT_IF_STATEMENT   // Add support for IF statement in rules (+4k2 code, -332 bytes mem)  
    #endif
#endif

#ifdef SENSORS
    #ifndef FIRMWARE_SENSORS
    #define FIRMWARE_SENSORS
    #endif
#endif

#ifdef LITE
    #ifndef FIRMWARE_LITE
    #define FIRMWARE_LITE
    #endif
#endif

#ifdef MIN
    #ifndef FIRMWARE_MINIMAL
    #define FIRMWARE_MINIMAL 
    #endif
#endif
```

where is my mistake?

barbudor, 2023-03-14:
A few problems
1) you originally tried to create your custom env by extending some tasmota variants. I see that you have now commented that. OK because that is not supported
2) you are ré-defining some FIRMWARE_XXX variants which does exactly what's it is intended to : make that build exactly that variant without anything more or less

Check again the First Rule of Compiling-your-own-build: You can only customize "tasmota" and not any of it's variant. As soon as you enable a variant, it overrides your customization.

If you want to do the like of a given variant you must include in your override.h all defines and undef needed.

In addition the name you are giving to your binary will create problems for upgrades especially on esp8266.
You should name your binaries "tasmota-whateveryouwantafterthedash.bin" but never change the "tasmota-" part.

Gifford47, 2023-03-15:
thanks for your explanation!

apart from the other builds, this build should work normally but it doesnt:

```
[env:tasmota_latest_sensors-shutter-full_rules]
;extends = env:tasmota-sensors
build_unflags           = ${common.build_unflags}
build_flags             = ${env.build_flags} -D FW_FULLRULES -D SHUTTERS -D SENSORS
```
i use a local docker container with the latest tasmota version

barbudor, 2023-03-15:
What do you mean should work ?

If your `user_config_override.h` is still the same, you are still doing the same errror

``` 
#ifdef SENSORS
    #ifndef FIRMWARE_SENSORS
    #define FIRMWARE_SENSORS
    #endif
#endif
```

means that ` -D SENSORS`  will define `FIRMWARE_SENSORS`  which will build `tasmota-sensors`

Stop using `FIRMWARE_XXXX`  in your user_config_override.h

blakadder, 2023-03-15:
This is not a docker-tasmota issue, if you need help compiling Tasmota join the Discord server

closed 2023-03-15 by blakadder