tasmota/docker-tasmota #24: Build failure

darrylb123, 2023-06-19:
This is my first attempt to compile Tasmota.  I believe I have followed the documentation.
I tried both grabbing the pre-built blakadder docker container and building my own. With the same result.

$ cd /var/tmp
$ git clone https://github.com/tasmota/docker-tasmota
$ cd docker-tasmota
$ docker build -t docker-tasmota .
$ cd ..
$ git clone https://github.com/arendst/Tasmota.git
$ docker run -ti --rm -v $(pwd)/Tasmota:/tasmota -u $UID:$GID docker-tasmota -e tasmota-minimal
Compiling Tasmota...
Usage: pio run [OPTIONS]
Try 'pio run -h' for help.

Error: Invalid value for '-d' / '--project-dir': Path '/tasmota' is not readable.
All done! Find your builds in Tasmota/build_output/firmware/

barbudor, 2023-06-19:
Are $UID and $GID variable correctly set on your system?

darrylb123, 2023-06-19:
UID was but GID wasn't. Didn't help though
$ docker run -ti --rm -v $(pwd)/Tasmota:/tasmota -u $(id -u):$(id -g)
docker-tasmota -e tasmota-minimal
Compiling Tasmota...
Usage: pio run [OPTIONS]
Try 'pio run -h' for help.

Error: Invalid value for '-d' / '--project-dir': Path '/tasmota' is not
readable.
All done! Find your builds in Tasmota/build_output/firmware/

blakadder, 2023-06-19:
this is a file permission issue... try building it in your users home dir first

barbudor, 2023-06-19:
I agree it is a persmission issue but works like a charm in /var/tmp too:

``` bash
barbu@mini-server:/var/tmp$ docker run -ti --rm -v $(pwd)/Tasmota:/tasmota -u $UID:$GID docker-tasmota -e tasmota-minimal
Compiling Tasmota...
Processing tasmota-minimal (platform: https://github.com/tasmota/platform-espressif8266/releases/download/2023.04.00/platform-espressif8266.zip; framework: arduino; board: esp8266_1M)
-------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing https://github.com/tasmota/platform-espressif8266/releases/download/2023.04.00/platform-espressif8266.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif8266@2023.4.0 has been installed!
Tool Manager: Installing platformio/toolchain-xtensa @ ~2.40802.0
Downloading  [####################################]  100%
```

I believe that he is not executing the docker with the same user that did the git clone

darrylb123, 2023-06-19:
> Platform Manager: ***@***.*** has been installed!

I ran  it out of /var/tmp.
If it's file permissions, it might be selinux. I run Fedora with selinux
enabled.
I'll try it later.

darrylb123, 2023-06-20:
And that was it.
setenforce 0
and it ran through.
It doesn't work in my home directory either, so it's something to do with
docker on Fedora
Thanks guys.

closed 2024-11-30 by Jason2866