tasmota/docker-tasmota #27: Building Tasmota32 fails

jeroenst, 2024-04-25:
When I try to build the development branch of Tasmota using Tasmota32 it fails:

```
docker run -i --rm -v $(pwd)/Tasmota:/tasmota -u $UID:$GID docker-tasmota -e tasmota32
Compiling Tasmota...
Processing tasmota32 (framework: arduino; platform: https://github.com/tasmota/platform-espressif32/releases/download/2024.04.13/platform-espressif32.zip; board: esp32)
--------------------------------------------------------------------------------
Platform Manager: Installing https://github.com/tasmota/platform-espressif32/releases/download/2024.04.13/platform-espressif32.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Platform Manager: espressif32@2024.4.13 has been installed!
Platform Manager: Removing espressif32 @ 2024.4.13
Platform Manager: espressif32@2024.4.13 has been removed!
IncompatiblePlatform: Development platform 'espressif32' is not compatible with PlatformIO Core v6.1.11 and depends on PlatformIO Core >=6.1.14.
```

Jason2866, 2024-06-03:
Please try again. Docker Image is updated.

bogd, 2024-11-30:
Same issue here, but with different versions:

```
Compiling Tasmota...
Processing tasmota32s3 (board: esp32s3-qio_qspi; framework: arduino; platform: https://github.com/tasmota/platform-espressif32/releases/download/2024.11.31/platform-espressif32.zip)
--------------------------------------------------------------------------------
Platform Manager: Installing https://github.com/tasmota/platform-espressif32/releases/download/2024.11.31/platform-espressif32.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@2024.11.31 has been installed!
Platform Manager: Removing espressif32 @ 2024.11.31
Platform Manager: espressif32@2024.11.31 has been removed!
IncompatiblePlatform: Development platform 'espressif32' is not compatible with PlatformIO Core v6.1.15 and depends on PlatformIO Core >=6.1.16.
```

Jason2866, 2024-11-30:
Docker image updated

closed 2024-11-30 by Jason2866