tasmota/docker-tasmota #29: firmware run error

HiconKong, 2024-10-14:
I want to build firmware by docker-tasmota and flash into SONOFF DUALR3 Lite.

I build tasmota fimrware from below steps
* git clone https://github.com/tasmota/docker-tasmota
* cd docker-tasmota && git clone https://github.com/arendst/Tasmota.git
* ./compile.sh tasmota
Then I got firmware at Tasmota/build_output/firmware/tasmota.bin. And I use Web Tool(https://tasmota.github.io/install) to install the firmware. But I got some error.
`
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
wrong chip id 0x8a90
wrong chip id 0x8a90
wrong chip id 0x8a90
wrong chip id 0x8a90
wrong chip id 0x8a90
wrong chip id 0x8a90
wrong chip id 0x8a90
ets Jul 29 2019 12:21:46
`

HiconKong, 2024-10-14:
[image]

barbudor, 2024-10-15:
tasmota.bin is for ESP8266
Sonoff Dual R3 Lite use an ESP32
You need to build tasmota32
Warning: For flashing an ESP32 using the webinstaller, you must use tasmota32.factory.bin (tasmota32.bin is only for OTA upgrade once you have already flashed tasmota32.factory.bin)

closed 2024-11-30 by Jason2866