tasmota/docker-tasmota #31: Compilation stops with error

kobuki, 2024-11-17:
I ran into a problem after successfully using the image for a few weeks. Now any compilation result claims that it's successful but actually images are not built. Log contents:

```
~/docker-tasmota$ cat docker-tasmota.log
Compiling Tasmota...
Processing tasmota32 (framework: arduino; platform: https://github.com/tasmota/platform-espressif32/releases/download/2024.11.31/platform-espressif32.zip; board: esp32)
--------------------------------------------------------------------------------
Platform Manager: Installing https://github.com/tasmota/platform-espressif32/releases/download/2024.11.31/platform-espressif32.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@2024.11.31 has been installed!
Platform Manager: Removing espressif32 @ 2024.11.31
Platform Manager: espressif32@2024.11.31 has been removed!
IncompatiblePlatform: Development platform 'espressif32' is not compatible with PlatformIO Core v6.1.15 and depends on PlatformIO Core >=6.1.16.

All done! Find your builds in Tasmota/build_output/firmware/
```

Console output:

```
~/docker-tasmota$ ./compile.sh tasmota32
From https://github.com/arendst/Tasmota
 * branch                development -> FETCH_HEAD
Checking Tasmota GitHub for the most recent development version

Running Docker Tasmota on Tasmota version development

Compiling builds:
tasmota32

Using your user_config_override.h and overwriting the existing file

✔ Finished!     Compilation log in docker-tasmota.log

cp: cannot stat '/home/user/docker-tasmota/Tasmota/build_output/firmware/tasmota32*': No such file or directory
WARNING:
Something went wrong while compiling tasmota32. Check compilation log
```

Fresh clone, latest image, no release or branch pinning. Note that on first run, specifying the build profile like `tasmota32` seems to be ignored. It wants to build all images instead. Stopping with ctrl-c and then restarting with the same command (`./compile.sh tasmota32`) does compile (or now, tries to start compiling) only the specified images.

Anyway, nice Docker image, it makes casual compilig Tasmota really easy.

barbudor, 2024-11-18:
The error message is clear: the platform io version in your docker image is obsolete and need to be update

Delete the current image and recreate an up to date one

To solve that in my automated daily build, I always delete the image after usage and recreate it every day

kobuki, 2024-11-18:
OK, I was thinking of building the image locally instead of pulling the latest anyway to see if it solves the issue. It seems like it's the expected way.

barbudor, 2024-11-18:
Either pull or build
But get an up to date one

closed 2024-11-18 by kobuki