tasmota/docker-tasmota #32: Reuse PlatformIO cache across runs?

FloRinke, 2025-02-08:
I noticed that each time I run the docker container, several large files are downloaded by platformio which would usually get cached. 
Is this intentional? (Maybe preserving the cache caused some issues later, which I'm not aware of at the moment?)

Otherwise I'd suggest persisting the platformio cachdir `/.platformio` outside the container.
I decided to mount in a directory via commandline using `-v $(pwd)/Tasmota/.docker_pio:/.platformio`, but it might be more suitable to define a named volume in the Dockerfile and let Docker handle storing it.

(I admit this was especially annoying in my case, because I had to rerun the build a few times in short succession in order to weed out some issues with my freshly created `user_config_override.h`. If you have an established config and only rebuild it for new Tasmota versions from time to time, the effect will be less severe.)

barbudor, 2025-02-08:
Are you sure ?

I though it was downloading only when you build the container image
I have decided to systematically delete the container image and rebuild in in my nighly build process to be sure that the latest platformio and sdk were used. In the past I had problems with the build failing because a newer version of platformio was required but not automatically updated.

I think it's up to how you are using the docker if you want to cache or not.
Your option is valid for repetitive builds

FloRinke, 2025-02-16:
Ah, I see what's going on: You are building for an ESP8266, whose toolchain is included in the docker image:
```
#8 22.79 PACKAGES: 
#8 22.79  - framework-arduinoespressif8266 @ 2.7.8 
#8 22.79  - tool-esptoolpy @ 4.7.6 
#8 22.79  - toolchain-xtensa @ 2.40802.200502 (4.8.2)
```

But when building for other targets (ESP32 and ESP32-S3 in my case), their respective tooling is not part of the container, and therefore downloaded on every run:

For ESP32
```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.1.1+sha.108a0792 
 - tool-esptoolpy @ 4.8.7 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
```

```
For ESP32-S3
PACKAGES: 
 - framework-arduinoespressif32 @ 3.1.1+sha.108a0792 
 - tool-esptoolpy @ 4.8.7 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-riscv32-esp @ 13.2.0+20240530 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
```

I don't think it makes any sense to try and include toolchains for all possible platforms, but providing some hint in the Readme could save some people without docker experience quite some time (both in buildtime and figuring out this behaviour).
I can provide a PR if you want.

Noschvie, 2025-03-10:
@FloRinke you will create a PR? Thanks!