tasmota/docker-tasmota #33: ERROR: Could not install packages due to an OSError

Noschvie, 2025-03-10:
Hello
command line `docker run -ti --rm -v ./Tasmota:/tasmota -u $UID:$GID docker-tasmota -e tasmota32`

what is going wrong if I get this error? Thanks

```
Compiling Tasmota...
Processing tasmota32 (framework: arduino; platform: https://github.com/tasmota/platform-espressif32/releases/download/2025.03.30/platform-espressif32.zip; board: esp32)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing https://github.com/tasmota/platform-espressif32/releases/download/2025.03.30/platform-espressif32.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@2025.3.30 has been installed!
Tool Manager: Installing https://github.com/tasmota/arduino-esp32/releases/download/v3.1.3.250302/framework-arduinoespressif32.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoespressif32@3.1.3+sha.be550b63 has been installed!
Tool Manager: Installing platformio/toolchain-xtensa-esp-elf @ 13.2.0+20240530
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-xtensa-esp-elf@13.2.0+20240530 has been installed!
Tool Manager: Installing platformio/tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-xtensa-esp-elf-gdb@14.2.0+20240403 has been installed!
Tool Manager: Installing platformio/tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-riscv32-esp-elf-gdb@14.2.0+20240403 has been installed!
Tool Manager: Installing https://github.com/tasmota/esptool/releases/download/v4.8.8/esptool.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@4.8.8 has been installed!
Tool Manager: Installing tasmota/tool-mklittlefs @ ^3.2.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-mklittlefs@3.2.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
Nothing more to solidify
-DCONFIG_TASMOTA_FLASHMODE_DIO
*** use provided user_config_override.h as planned ***
*** use provided platformio_override.ini as planned ***
*** use provided platformio_tasmota_cenv.ini as planned ***
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32.html
PLATFORM: Espressif 32 (2025.3.30) > Espressif Generic ESP32 >= 4M Flash PSRAM, Tasmota 2880k Code/OTA, 320k FS
HARDWARE: ESP32 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.1.3+sha.be550b63
 - tool-esptoolpy @ 4.8.8
 - tool-mklittlefs @ 3.2.0
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Converting tasmota.ino
Installing Arduino Python dependencies
Collecting rich-click>=1.8.6
  Downloading rich_click-1.8.8-py3-none-any.whl.metadata (7.9 kB)
Collecting tasmota-metrics>=0.4.3
  Downloading tasmota_metrics-0.4.3-py3-none-any.whl.metadata (1.4 kB)
Requirement already satisfied: click>=7 in /usr/local/lib/python3.13/site-packages (from rich-click>=1.8.6) (8.1.8)
Collecting rich>=10.7 (from rich-click>=1.8.6)
  Downloading rich-13.9.4-py3-none-any.whl.metadata (18 kB)
Collecting typing_extensions>=4 (from rich-click>=1.8.6)
  Downloading typing_extensions-4.12.2-py3-none-any.whl.metadata (3.0 kB)
Collecting pyyaml (from tasmota-metrics>=0.4.3)
  Downloading PyYAML-6.0.2-cp313-cp313-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (2.1 kB)
Collecting markdown-it-py>=2.2.0 (from rich>=10.7->rich-click>=1.8.6)
  Downloading markdown_it_py-3.0.0-py3-none-any.whl.metadata (6.9 kB)
Collecting pygments<3.0.0,>=2.13.0 (from rich>=10.7->rich-click>=1.8.6)
  Downloading pygments-2.19.1-py3-none-any.whl.metadata (2.5 kB)
Collecting mdurl~=0.1 (from markdown-it-py>=2.2.0->rich>=10.7->rich-click>=1.8.6)
  Downloading mdurl-0.1.2-py3-none-any.whl.metadata (1.6 kB)
Downloading rich_click-1.8.8-py3-none-any.whl (35 kB)
Downloading tasmota_metrics-0.4.3-py3-none-any.whl (23 kB)
Downloading rich-13.9.4-py3-none-any.whl (242 kB)
Downloading typing_extensions-4.12.2-py3-none-any.whl (37 kB)
Downloading PyYAML-6.0.2-cp313-cp313-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (733 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 733.4/733.4 kB 14.1 MB/s eta 0:00:00
Downloading markdown_it_py-3.0.0-py3-none-any.whl (87 kB)
Downloading pygments-2.19.1-py3-none-any.whl (1.2 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 11.0 MB/s eta 0:00:00
Downloading mdurl-0.1.2-py3-none-any.whl (10.0 kB)
Installing collected packages: typing_extensions, pyyaml, pygments, mdurl, tasmota-metrics, markdown-it-py, rich, rich-click
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: '/usr/local/bin/pygmentize'
Consider using the `--user` option or check the permissions.

*** Error 1
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 128 compatible libraries
Scanning dependencies...
```

Jason2866, 2025-06-13:
@Noschvie Try latest version. CI run was successful, should be fixed.

closed 2025-06-13 by Jason2866

Noschvie, 2025-06-15:
Confirmed. Thanks again!