tasmota/docker-tasmota #34: USE_STABLE=1 on Mac OS is not working

tzahari, 2025-03-23:
I tried to compile Tasmota with the latest stable versions on Mac OS X 15.3.2

But it failed because the grep -P parameter is not available on Mac OS:
https://github.com/tasmota/docker-tasmota/blob/0f4be2568eee488a40ceafa673d5200882230d9d/compile.sh#L28

```bash
export USE_STABLE=1
./compile.sh tasmota

Checking Tasmota GitHub for the most recent release version
grep: invalid option -- P
usage: grep [-abcdDEFGHhIiJLlMmnOopqRSsUVvwXxZz] [-A num] [-B num] [-C[num]]
        [-e pattern] [-f file] [--binary-files=value] [--color=when]
        [--context[=num]] [--directories=action] [--label] [--line-buffered]
        [--null] [pattern] [file ...]
Failed to fetch/set Tasmota branch! Check internet connection and try again.
```

If I also set `export TASMOTA_BRANCH=v14.5.0` then the build works.

tzahari, 2025-03-23:
I found this solution on https://stackoverflow.com/a/75833940/23039308 and this is working on Mac OS X

```bash
wget -qO - https://api.github.com/repos/arendst/Tasmota/releases/latest | grep -i "tag_name" | awk -F '"' '{print $4}'
v14.5.0
```

Jason2866, 2025-06-13:
Try latest, should work on Linux and Mac

closed 2025-06-13 by Jason2866